Ultimaker/Cura #18105: Slicing fail

crankysaint, 2024-01-23:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3

### Describe your model

The model sliced successfully scaled down to 25 percent previously.

### Add your .zip here ⬇️

[Uploading Dalek_thing.zip…]()

Asterchades, 2024-01-23:
First things first, you didn't let your upload finish. As a result there's no actual project file attached to your post. You need to wait until the text changes a second time to indicate a successful upload.

Now, you may want to try again in the [just-released 5.7 alpha](https://github.com/Ultimaker/Cura/releases/tag/5.7.0-alpha.0). It includes some fixes with regards to faults such as this, but (perhaps more importantly) it also includes new logging facilities to better isolate the cause of crashes. This is an extremely early build so it's not recommended to use it as your "daily driver" but the two versions can co-exist on the one system.

crankysaint, 2024-01-23:
Woops. Never done this before. Re-uploaded the file. I'll give 5.7 a shot.