Ultimaker/Cura #18115: Slicing failed. Model too big?

truszko1, 2024-01-24:
### Cura Version

5.6.0

### Operating System

Windows 11, x64

### Printer

Creality Ender-3 V3 SE

### Name abnormal settings

Lightning infill pattern, 10% infill

### Describe your model

Western USA. It's a part of a world map.

### Add your .zip here ⬇️

Western USA, a part of a Topographic World Map set of models from thingiverse: https://www.thingiverse.com/thing:6409783/files

Slicing starts, I can hear my computer fan go off, CPU spikes. After maybe 5 min, slicing stops and I got a popup asking to report a bug. And here we are.

[CE3V3SE_6_USA_1.zip](https://github.com/Ultimaker/Cura/files/14031654/CE3V3SE_6_USA_1.zip)

GregValiant, 2024-01-26:
I sliced this with no problem on my Win 10 Pro laptop.  This might be a problem/bug associated with Win11.  @5axes is keeping a list of bug reports that seem to be indicating an incompatibility with Win11.

That model does have some errors in it.  That might be a problem as well, but as I say, it did slice for me in 5.6.0.
[image]

GregValiant, 2024-01-26:
If you can duplicate the problem then use the "Help | Show Configuration" folder command.  In that folder will be the "Cura.log" file.  Zip that and post the zip folder.
In the last couple of months there have been a lot of reports regarding Cura installed on Win11 systems and a lot of models that won't slice.  Then I check them on my Win 10 system and they slice with no problem.  The Cura team is aware that there is a problem someplace but I don't know how much (if any) progress is being made trying to chase it down.

truszko1, 2024-01-26:
I can reproduce it. I’ll upload the logs! Hopefully that helps.