Ultimaker/Cura #18124: ERROR SLICING

Giovasaurio, 2024-01-25:
### Cura Version

5.5

### Operating System

Windows 11

### Printer

FLSun Super Racer

### Name abnormal settings

Scale 150 mm in Z

### Describe your model

Guinivere, character of mobile game MOBILE LEGENDS BANG BANG

### Add your .zip here ⬇️

[error.zip](https://github.com/Ultimaker/Cura/files/14045415/error.zip)

Giovasaurio, 2024-01-25:
The model scaled at 100% slincing correctly, but when increasing its scale (186%) it gives me an error in the slincing

nallath, 2024-01-25:
Could you also slice it with this build: https://github.com/Ultimaker/Cura/discussions/18080? This build has extra logging to provide us with more info.

Giovasaurio, 2024-01-26:
I am somewhat inexperienced in the subject of 3D printing, I do not know how to apply the settings you suggest. :c