Ultimaker/Cura #18125: Anycubic Kobra Max drags nozzle on printed parts - but not print bed.

spartyxii, 2024-01-25:
### Cura Version

5.5.0 and 5.6.0

### Operating System

Windows 10 and MacOS

### Printer

Anycubic Kobra Max

### Reproduction steps

1. Select files in Cura and orient them properly
2. Change retraction rate and retraction speed to reduce stringing issues
3. Slice files
4. Autolevel AKMAX bed
5. Begin print
6. Once the first layer is printed, every Z hop motion leaves a scoured mark across the print. This continues through all layers of the print.

Additional Notes:
     I don't know whether this is an issue with my particular printer, or with Cura interacting with my printer. I've tried this on my girlfriend's Kobra Max, it is identical to mine and suffers the same issue.
Things I have already tried to fix this issue:
- Reducing flow rate to prevent over extrusion
- increasing initial layer height
- Increasing Z Hop Height
- Increasing Z Hop Speed
- Decreasing travel speed
- Decreasing initial travel speed
- Decreasing Travel Acceleration
- Decreasing Travel Jerk
- Installing the Z Offset Plugin
- Increasing the Z Offset to .02 and then .03

Despite all this tinkering, I have not made any improvement in the nozzle's determined gouging of the printed pieces. I am planning some major print work soon, and I am afraid to print anything large for fears of it coming loose under the pressure of the nozzle, or worse damaging the hotend in some way.

### Actual results

Print is scratched, grooved, and potentially knocked free of the heated bed due to dragging of the nozzle.

### Expected results

Print should be without scratches/grooves.

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2024-01-26:
You haven't specifically mentioned it, nor included a Project File so I can check, but take a look at your Combing settings. Any Combing movement doesn't count as a retraction, so will not trigger a Z Hop and can easily result in scarring.

It's also possible (though I feel unlikely) that you could be running into the retraction limits as set by `Retraction Minimum Travel`, `Maximum Retraction Count`, and/or `Minimum Extrusion Distance Window`. The default values for these shouldn't be coming into play in a situation like this, but without any way to tell what values you're using it may still be worth your checking.

spartyxii, 2024-01-27:
Asterchades,

I didn't include a project file originally because this issue is happening with every file that I try to print, but I didn't consider the usefulness of having a project file for settings investigation. I've attached one to this comment.

Additionally, as per your advice, I checked my combing settings and found that combing is off entirely. Is this favorable for my troubleshooting process? Thank you.

I'm unsure as to what the other values you mentioned should be, so I would appreciate if you wouldn't mind examining the settings in this project file for them? Thank you. (Had to zip it as the project file format is not supported for posting here).
[spartyxiitestprojectfile.zip](https://github.com/Ultimaker/Cura/files/14070727/spartyxiitestprojectfile.zip)