Ultimaker/Cura #18127: slicing fails Cura 5.6 Kingroon KLP1 Cache

paulgiordano, 2024-01-25:
### Cura Version

5.6

### Operating System

Linux

### Printer

Kingroon KLP1 Klipper

### Reproduction steps

selected printer, add an STL, slice, and get the error

### Actual results

Error message as on screen shot

### Expected results

A normal slice

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/14049353/cura.log)
[HMG7.4 Creality Spider 3 Pro Mount V6.zip](https://github.com/Ultimaker/Cura/files/14049376/HMG7.4.Creality.Spider.3.Pro.Mount.V6.zip)
[image]

paulgiordano, 2024-01-25:
Tie this to Tree supports, everywhere, overhang 50, zigzag, 0.0 density. Actually it looks like tree support is broken in this instance, no matter what the other parameters are.

nallath, 2024-01-25:
Could you try to use this build: https://github.com/Ultimaker/Cura/discussions/18080? 

If that doesn't fix it, we at least get more information about what is going wrong.

paulgiordano, 2024-01-25:
That fixes it for me. Thanks for being so quick!

closed 2024-01-25 by paulgiordano