Ultimaker/Cura #18133: Slicing failed when using tree supports

mebeggs, 2024-01-26:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Neptune 3 pro

### Name abnormal settings

All defaults except for selecting tree supports.  It slices ok with defaults, and with normal supports.

### Describe model location

no, neither

### Describe your model

Yes it slices ok without tree supports.  This is part of a popular public free model. no haven't tried doing a fix.

### Add your .zip here ⬇️

[Hub-error.zip](https://github.com/Ultimaker/Cura/files/14059643/Hub-error.zip)

mebeggs, 2024-01-26:
More info...  I just started 3d printing a month ago.  I think I've been using the same version of Cura the whole time.  This slicing problem just started maybe a week ago.  I've been using tree supports successfully up to this point.  In fact until this issue arose, I never used "normal" supports.

ThomasRahm, 2024-01-26:
There is an alpha that contains fixes for tree support crashes here: https://github.com/Ultimaker/Cura/discussions/18080