Ultimaker/Cura #18140: Large supports and can't go back to normal size

WinterLFG, 2024-01-26:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Modded Ender 3 v2

### Reproduction steps

Turned off supports and turned them back on after uploading a different file.

### Actual results

Weird super large supports. No idea how it happened but for whatever reason the supports for only normal supports are just massive and thick, even with size settings set to their lowest it just creates the same thick supports with a small bevel at the top.

### Expected results

Should change the size of normal supports, but stays large.

### Add your .zip and screenshots here ⬇️

[CE3E3V2_46 silver.zip](https://github.com/Ultimaker/Cura/files/14065907/CE3E3V2_46.silver.zip)
[image]
[image]

The first image is what it looks like with the default setting of 0.6 on the "Support Horizonal expansion" and the second is with it at 0.

GregValiant, 2024-01-26:
Thanks for the report.
With the settings in your project file I can't generate any supports.  After lowering the "Minimum Support Area" to 0.5 I get this.
(Looks good to me.)
[image]

If I enable "Use Towers" it looks more like what you have there.
[image]

So check if you have "Use Towers" enabled and turn it off.  Let us know if that works for you.  I'll remove the bug label for now.

WinterLFG, 2024-01-26:
Thanks for the info, for whatever reason towers turned on. I turned it off and it, kinda? fixed it, maybe? My supports still aren't clean like yours and are quite jagged, it also just gave up on the idea of supporting the circles all together with towers off. (I also had to put horizonal expansion at 0.1 otherwise the support wouldn't generate at all.
[image]

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant