Ultimaker/Cura #18146: Initial Layer Speed Partially Incorrect

wulfhednar02, 2024-01-27:
### Cura Version

5.6.0

### Operating System

Kubuntu Linux

### Printer

Geeetech Mizar Pro 0.4mm nozzle

### Reproduction steps

1. Set all print speeds to 40mm/s
2. Set Initial Layer Speed to 10mm/s
3. Slice

### Actual results

Outer wall of the initial layer is 40mm/s.

Note: The model is not watertight and is dropped below the build plate level by -0.1mm

### Expected results

Expected all of the initial layer to be 10mm/s

### Add your .zip and screenshots here ⬇️

[dragon.zip](https://github.com/Ultimaker/Cura/files/14070414/dragon.zip)
[image]
[image]

wulfhednar02, 2024-01-27:
I have also been experiencing issues with retraction being set but not being implemented in the gcode. The original and retraction problems go away when ever I make the values permanent values. The issue stated and the retraction become wrong whenever I set the settings to "Use material setting".

GregValiant, 2024-01-28:
Thanks for the report.
The Initial layer speed problem has been reported previously (#17602 ).
That model has a lot of errors.  I would not trust it to slice correctly.
[image]

In regards to the "Retraction" settings, I can't duplicate that.  When I click on "Use value from material" the current setting is correctly over-ridden by the setting of the current material "Transparent PETG" with a retraction speed of 80.

wulfhednar02, 2024-01-28:
`I would not trust it to slice correctly.`
For sure.

The retraction settings always look correct. It was not until I actually printed the model that the I "see" the retraction is missing. Could be this model, but once I put the numbers back to manually entered, the retraction came back for this model.

GregValiant, 2024-01-29:
When I slice the project (after repairing the model) there are 3475 retractions in the gcode (and a corresponding 3475 primes).
With the retraction distance of 1mm the extruder doesn't rotate very far to make the retraction.  It looks to be about 11° of rotation per mm of filament.  That would be pretty hard to see.
Post one of the gcode files that you feel is in error.  I have some analysis tools I can run it through.

I'll mark this as a duplciate report regarding the speed problem.  If the retraction issue seems like a bug it should go on a separate report.

wulfhednar02, 2024-01-29:
Thank you. I have everything working at this moment, nothing seems to be lacking. I would say just close this issue and I will get more information if I run across the retraction thing again. I think it is a thing, but there is no telling where it could be if I do not have information for you to track it down. :stuck_out_tongue: I know what that is like...

GregValiant, 2024-01-29:
I'll leave the "duplicate bug report" on this and go ahead and close it then.  Glad to hear things are working.

closed 2024-01-29 by GregValiant