Ultimaker/Cura #18151: Elegoo Cura 5.6 Slice fail

akacary, 2024-01-27:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Neptune 4 Max

### Name abnormal settings

No

### Describe model location

No

### Describe your model

Yes on older Cura i have no problems but with this new version i cant slice anything ive printed before

### Add your .zip here ⬇️

Cura 5.6.0 just doesn't work regardless of plugins, modles, rotation or size
[Print Project.zip](https://github.com/Ultimaker/Cura/files/14073020/Print.Project.zip)

nallath, 2024-01-31:
Could you try it again with this build: https://github.com/Ultimaker/Cura/discussions/18080 It doesn't contain a fix, but it has a lot more logging options, which will help us figure out what is going wrong

akacary, 2024-01-31:
There isn't a profile for neptune 4 Max but it's slicing if i use profile Neptune 3 Max, Neptune 4 and Neptune 4 Pro profile  so its working if you can add the 4 Max profile that would be grate.