Ultimaker/Cura #18155: Slicing error when i use Hueforge stl

Damada42, 2024-01-27:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Sidewinder X1

### Name abnormal settings

i always change the layer height and the base layer height

### Describe model location

no

### Describe your model

no

### Add your .zip here ⬇️

[ps01_Front_179x255_cura560.zip](https://github.com/Ultimaker/Cura/files/14073661/ps01_Front_179x255_cura560.zip)
it was a stl generate from hueforge and the most models from hueforge can not slice with Cura...
I also try it without filamentchange and with "normal" settings, so 0,3mm base layer and other layer hight 0,2 and 10% infill but also with this settings i get an error

nallath, 2024-01-31:
Could you try it again with this build: https://github.com/Ultimaker/Cura/discussions/18080 It doesn't contain a fix, but it has a lot more logging options, which will help us figure out what is going wrong

Damada42, 2024-02-01:
[ps01_Front_179x255.stlcura570Alpha.zip](https://github.com/Ultimaker/Cura/files/14123475/ps01_Front_179x255.stlcura570Alpha.zip)
I have try it out with the alpha version but der is the same problem.
But when i set layerheight to 0,1 or higher then it works.
Where can i find the logging option or loggingfile?

nallath, 2024-02-01:
It should have automatically reported slicing crashes to us, together with information where in the code it fails.