Ultimaker/Cura #18165: Weird Visual Glitch

nebebbeubebebe, 2024-01-28:
### Cura Version

5.4.0

### Operating System

MacOS

### Printer

Ender 3 S1

### Reproduction steps

I just sliced a model

### Actual results

[image]
It bugged out

### Expected results

It should have just been the sliced model

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-01-29:
Thanks for the report.
This bug comes up about once a year.  I call it the "Peter Maxx" bug.  So far it's been impossible to duplicate on purpose.  It probably has something to do with the video drivers or OpenGL.  On the other examples the generated gcode was fine.
That one of yours is pretty cool so I'll add it to my collection.

This one showed up on my system one time a couple of years ago.  I think it was with Cura 4.10 so the bug has been around for a while.  It's really good at hiding.
[image]

I don't know if the Cura team has ever made any progress on this.  Since it can't be duplicated on purpose it isn't likely to be fixed.
BTW Your project upload seems to have a glitch in the link.  Instead of downloading a file it opens a new instance of this page.

fieldOfView, 2024-01-29:
It is unfortunately a common visualisation issue on Mac OS. You could try setting the option to "Force layer view compatibility mode" in the General preferences in Cura. This will use a less-fancy (and feature-limited) version of OpenGL.

What you are seeing is purely a visualization issue, it will not affect printing. Here's what's going on (copy/pasta from another similar question some time ago):

Cura is a cross-platform application. It is written to use frameworks and APIs that are available on Windows, Mac OS and on Linux. There are some minor bits of platform-specific code in Cura, but most of it is the same between those three operating systems. This - by the way - is also why Cura has this non-standard application menu on top of its window, but I digress.

A while ago, Apple unilaterally decided that one of those well-established cross-platform frameworks, OpenGL, was not where they wanted to put their time and effort in, and instead started to push Metal as the API to use for 3d graphics. That API is only available on Mac OS (and its derivatives such as IOS, Ipad OS, etc), so it can not be used on all three platforms. OpenGL is still available on Mac OS, but it is no longer actively maintained. What this means is that if some change in Mac OS triggers a bug in their implementation of OpenGL, they will not fix it. What you see on your system is a visual glitch caused by such a bug in their OpenGL implementation. Reinstalling Cura will not fix that bug.

Ultimaker could fix this in three ways:

    Try to work around this issue until something else breaks, ad infinitum

    Use OpenGL on Windows and Linux, and totally rewrite graphics for Mac OS specifically

    Wait until a new cross-platform framework matures that under the hood uses Metal on Mac OS, and eg Vulkan on Windows and Linux, in a way that platform-specific code is not needed

The second option is prohibitively expensive for the number of Mac OS user vs Windows (and Linux) users that pay the bills at Ultimaker by buying printers. Not just for the cost of development once, but also for maintenance of two separate codebases. Not going to happen.

The third option, in addition to being dependent on what emerges - which is a slow process - also has the disadvantage that this new framework likely excludes a lot of older computers, graphics cards and operating system versions. But ultimately, this is probably the way forward.