Ultimaker/Cura #18166: Is CURA still not working in Windows 11 Pro?

hwangomin, 2024-01-29:
### Cura Version

5.6.0 / 5.4.0

### Operating System

Windows11pro

### Printer

Creality 3D Ender-3 V2

### Reproduction steps

Description: I installed the computer operating system as Windows 11 Pro and proceeded with slicing by installing CURA version 5.6.0. However, no matter what gcod I input on my 3D printer, it only creates a triangle in the bottom left corner of the bad.

Is CURA still not working in Windows 11 Pro?

### Actual results

Filament is output in a triangular shape only in the lower left part of the bed.

### Expected results

Force power off.

### Add your .zip and screenshots here ⬇️

사진자료 첨부하겠다

Asterchades, 2024-01-29:
I've been using Cura under Windows 11 Pro from about two weeks after Win11 originally went RTM back in late 2021. This would be with versions ranging from 4.12.0 up to the current 5.6.0, including multiple pre-release versions. I can honestly say I've never experienced what you're reporting, nor have I seen any other reports of the same.

More information is going to be required to determine what, if anything can be done to assist you. The Cura log file you were asked to include when making this report will contain a basic read-out of what hardware you're using to see if there's some kind of incompatibility, and a project file (which you were also asked to include) may also be of assistance. In this particular case there's possibly also useful information in the GCode output as well, which would be better having been retrieved from your SD card rather than a more reliable storage medium.

hwangomin, 2024-01-29:
p{margin-top:0;margin-bottom:0}
    

    Thank you for your quick and kind response.--------- 원본 메일 ---------

  
    보낸사람: Asterchades ***@***.***>
    받는사람: Ultimaker/Cura ***@***.***>
      참조: hwangomin ***@***.***>,Author ***@***.***>
    날짜: 24.01.29 15:40 GMT +0900
    
    제목: Re: [Ultimaker/Cura] Is CURA still not working in Windows 11 Pro? (Issue #18166)
    
    
  
  
    
I've been using Cura under Windows 11 Pro from about two weeks after Win11 originally went RTM back in late 2021. This would be with versions ranging from 4.12.0 up to the current 5.6.0, including multiple pre-release versions. I can honestly say I've never experienced what you're reporting, nor have I seen any other reports of the same.
More information is going to be required to determine what, if anything can be done to assist you. The Cura log file you were asked to include when making this report will contain a basic read-out of what hardware you're using to see if there's some kind of incompatibility, and a project file (which you were also asked to include) may also be of assistance. In this particular case there's possibly also useful information in the GCode output as well, which would be better having been retrieved from your SD card rather than a more reliable storage medium.