Ultimaker/Cura #18167: Adding a profile for the neptune 4 0.2, 0.6 and 0.8 nozzles

callimburg, 2024-01-29:
### Is your feature request related to a problem?

I'd just like to print with differents nozzle, but the profile ends up as "not supported"

### Describe the solution you'd like

I'd like to know if there's at least a plan to create those profiles

### Describe alternatives you've considered

Setting a guide on how to adapt the 0.4 profile to different nozzles side

### Affected users and/or printers

Neptune 4 series

GregValiant, 2024-01-29:
The definition files for non-UltiMaker printers necessarily come from members of the community (owners of the machines) or from the manufacturer.
I don't know your printer but you should be able to use whatever nozzle you want in that printer.  Just set the layer height and line width in Cura so it is compatible with the actual nozzle that is installed.

callimburg, 2024-01-29:
Thanks, I forgot to put the model of my printer in the post but it's a Neptune 4 Pro, I mean I understand that line height and line width but what about the extrusion? I haven't found a single soul capable of telling me the settings to watch for a nozzle change

GregValiant, 2024-01-29:
I switch nozzles all the time.  The only things you need to keep an eye on are Line Width, and the Layer Height.  Print speed becomes involved if you are trying to print with a 0.8 or a 1.0 because the flow rate gets very high and can outstrip how much melted material the printer can put out.  Higher flow rates can also effect the hot end temperature because as the plastic flows through the nozzle it cools it off.
You can push a 0.4 nozzle to .6 line width with no problem.  Going the other way - trying to print a .3 line width with a 0.6 nozzle doesn't work because there just isn't any squish.
Just keep your numbers reasonable for whatever size nozzle you have installed and you shouldn't have any trouble.  The extrusion volume is always "Line Width * Layer Height * Length of Extrusion"

The Neptune 4 series has been added to Cura.  It should show up in the 5.7 release.  It includes 2,4,6,8, nozzle sizes.

callimburg, 2024-02-04:
Thanks for the tips but editing the width of the layer isn't enough to adapt the profiles, hoping to see the new profiles for the nozzles since they don't appear on the alpha version

GregValiant, 2024-04-18:
The problem with the Neptune was that the printer was new.  I see that in Cura 5.7.0 the nozzle files have been added.
I'll go ahead and close this as the issue looks to have been resolved.

@wgillispie  In the case of the Longer, the definition files submitted to UltiMaker for their printers did not include additional Nozzle sizes or Quality files.  So this problem was created by Longer as they authored the definition files.
I'll go ahead and close this as UltiMaker can't be responsible for someone else's product.  I hope you understand.

Because the Longer definition already shows that the printer has "variant nozzles", you can add nozzle files and quality files and it will work.
The quality files are in - "C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\**quality**\longer"
The nozzle files are in - "C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\**variants**\longer"
The files are simple text files.  Open them in a text editor, change one or two lines, and do a "save-as" using the same file naming convention.  Adding the files can be done in a few minutes.  It's repetitious but not difficult.

wgillispie, 2024-04-18:
Thank you for your fast response I will try to add them like you said once
again thank you very much

micnorian14, 2025-03-29:
There are still no quality profiles for any nozzle size profiles other than 0.4. Among many things that need fine tuning, you absolutely must change the retraction distance from 0.4 to something like 0.8. I am still fine tuning mine. 

Whats more - the printer start G-code has a line to preheat the head to 140C before setting it to whatever is in the print file's G-code. This appears to fail using the latest firmware for the Neptune 4. Instead the printer will attempt to print at 140C regardless of what you set in the slicer. I suggest either deleting the preheat line or changing it to something like what I have done. 

[image]

GregValiant, 2025-03-30:
@micnorian14  Please read the previous posts.  I have absolutely no idea what a Longer or Neptune 4 looks like much less how they need to be set up so the gcode works with their firmware.  It is up to users of those machines to make the necessary changes to the printer definition files and submit them to UltiMaker as a Pull Request.