Ultimaker/Cura #18171: Slicing Failed

farfenkugell, 2024-01-29:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Creality Ender-3 V3 SE

### Describe your model

Other parts of this model work

### Add your .zip here ⬇️

[CE3V3SE_MHW_Odogo_T_H.zip](https://github.com/Ultimaker/Cura/files/14086903/CE3V3SE_MHW_Odogo_T_H.zip)

https://www.thingiverse.com/thing:3936411

nallath, 2024-01-31:
Could you try it again with this build: https://github.com/Ultimaker/Cura/discussions/18080 It doesn't contain a fix, but it has a lot more logging options, which will help us figure out what is going wrong

farfenkugell, 2024-01-31:
Weirdly enough it does work with that build. 
[image]