Ultimaker/Cura #18182: Unable to drop Model Level on the bed

ravo44, 2024-01-30:
### Cura Version

5.6.0

### Operating System

Win10 

### Printer

Elegoo Neptune 4 Plus

### Reproduction steps

1. Cura 5.6 gives error when you drop model on plate.
2. When you look underneath you see that it is not fully down.

[beta bracket.zip](https://github.com/Ultimaker/Cura/files/14105049/beta.bracket.zip)
Have no idea why it does this..

### Actual results

3.Model Underneath the Plate showsDifferent colour's red and blue .
4.Model Checked and is level and square.

### Expected results

Expected results
Model Should be colour blue underneath the Plate.
[image]

### Add your .zip and screenshots here ⬇️

[Bed Plate Error.zip](https://github.com/Ultimaker/Cura/files/14105109/Bed.Plate.Error.zip)

ravo44, 2024-01-31:
Problem Solved 
Cad drawing error..
sorry guys :))

GregValiant, 2024-02-01:
I didn't have any trouble with it.  It just wasn't laying flat.  One of the rotation tools is "Lay Flat" and it fixed the problem.  I'll go ahead and close this.
[image]

closed 2024-02-01 by GregValiant