Ultimaker/Cura #18187: Échec de la découpe avec une erreur inattendue. Signalez un bug sur notre outil de suivi des problèmes.

jgojva, 2024-01-31:
### Cura Version

5.5.0 beta 1

### Operating System

Windows 11

### Printer

Elegoo neptune 3 plus / Klipper

### Reproduction steps

le problème deviens récurant

### Actual results

c'est la 2eme pièces travaillé avec 3D Builder qui ne peut pas être slicer

### Expected results

je ne vois pas ou est l'erreur

### Add your .zip and screenshots here ⬇️

[madmoto v2 .zip](https://github.com/Ultimaker/Cura/files/14109341/madmoto.v2.zip)

5axes, 2024-01-31:
Pas de soucis pour moi .
[image]
Un problème avec tes réglages mais sans fichier projet impossible à dire.

jgojva, 2024-01-31:
Merci beaucoup je vais creuser pour trouver mon souci !Envoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : 5axes ***@***.***> Date : 31/01/2024  18:42  (GMT+01:00) À : Ultimaker/Cura ***@***.***> Cc : jgojva ***@***.***>, Author ***@***.***> Objet : Re: [Ultimaker/Cura] Échec de la découpe avec une erreur inattendue. Signalez un bug sur notre outil de suivi des problèmes. (Issue #18187) 
Pas de soucis pour moi .
image.png (view on web)
Un problème avec tes réglages mais sans fichier projet impossible à dire.