Ultimaker/Cura #18192: Can't Sslice when orientation changed, can't save if orientation is left alone

JayJay2105, 2024-01-31:
### Cura Version

5.4.0

### Operating System

Mac OS Sonoma 14.2.1

### Printer

Reality CR10-S4

### Reproduction steps

Loaded stl - produced in Fusion 360, Sliced ok - can't save to removeable drive - creates file 0 bytes, save to HDD but file has errors if trying to move to removable sd card.
Tried re orienting model - won't slice

### Actual results

Slice fails - Cannot save to relabel drive, save to hdd - file is corrupted if trying to move to sd card

### Expected results

Sliced and be able to save to removable drive (sd card)

### Add your .zip and screenshots here ⬇️

[CCR1_Ev Charger Socket Swivel v1.3mf.zip](https://github.com/Ultimaker/Cura/files/14115314/CCR1_Ev.Charger.Socket.Swivel.v1.3mf.zip)

Vandresc, 2024-02-02:
HI @JayJay2105 , we addressed some of the save options in 5.6, could you please try in the latest Cura 5.6 and let us know if the issue persists (saving part)? If that is the case please also upload a copy of your logs, thank you! 
Could you try with another SD card or if you don't have access to one, maybe try another gcode file or another file and see if that file is also corrupted when you save it?

JayJay2105, 2024-02-02:
Thank you for your response, I have had a friend slice and save the file, and that all worked well, have tried a different sD and that worked! - I will bin the bad sd card.

Thank you for looking into this for me. Keep up the great work!