Ultimaker/Cura #18193: No extrude when the line with is diferent

MarLaf1975, 2024-01-31:
### Cura Version

5.5

### Operating System

windows 10

### Printer

Prusa clone

### Reproduction steps

1, in a 0.4 nozle i define a 0.5 or 0.38 with of line 
2, i define a  80% flow
3. Generate G-code
4. Print
5. The extruder motor is unfuntional
6. Return to 0.4 line and the motor of extruder run

### Actual results

The motor of extrude not run

### Expected results

The motor of extruder must run

### Add your .zip and screenshots here ⬇️

[CFFFP_Extrusion_Test.zip](https://github.com/Ultimaker/Cura/files/14118187/CFFFP_Extrusion_Test.zip)

GregValiant, 2024-02-01:
Thanks for the report.
In your settings you have the line width at 0.50 and that's fine.  You have the "Flows" set to 0.8%.  Your extruder is probably turning but at less than 1% flow it isn't going to turn very fast.

All-in-all you have some strange settings.  Your inner wall print speed is higher than your travel speed.  Your retraction speed of 10mm/sec is really low and your infill print speed is really high.  When you bounce speeds around for each feature it takes a while for the molten plastic to react inside the nozzle and the pressure never settles down.  That makes it very hard to tune the printer.

I don't see a bug here so I'll remove the label.  Your settings need to be adjusted towards normalcy.  
This is an analysis of the gcode I generated from your project:
**Line Width calculated from E values, layer height, and extrusion length**
0.12 Widest
0.00 Average
0.00 Narrowest

**Flow Rate in mm³/sec calculated from print speed and E values**
0.08 Average
0.14 High
0.03 Low

A normal average would be 5 to 10mm³/sec so you are at less than 1% of what would be normal.

[image]

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant