Ultimaker/Cura #18203: Curved surface decrease speed randomly

AnthonyTTaylor, 2024-02-01:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 5 Plus

### Reproduction steps

1)Load model with curved surface.
2)Slice model with any settings (0.8 default is what I used)
3)Analyze Model or Print

### Actual results

Curved surfaces will randomly slow down print speed causing zips and bulge on the surface. 

### Expected results

Smooth prints.

### Add your .zip and screenshots here ⬇️

[CE5P_part2_issue.zip](https://github.com/Ultimaker/Cura/files/14131279/CE5P_part2_i
[image]
ssue.zip)

GregValiant, 2024-02-01:
Thanks for the report.
That preview is in error.
This is the Cura slice with my speed checker post processor running:
[image]

I read the file into MS Excel
**Initial Layer Feed rate**
39.83 Average
45.00 High
26.00 Low
**All other layers Feed Rate**
50.93 Average
60.00 High
32.80 Low
**Layer 9 only Feed Rates**
52.38 Average
60.00 High
39.00 Low

This is layer 9 of the gcode file sliced by Cura and viewed in the PrusaSlicer viewer.
[image]

And this is layer 9 of the speed view in Cura.
[image]

This is the Speed view in IdeaMaker:
[image]

So PrusaSlicer, Cura, IdeaMaker and Microsoft Excel all agree on the speeds and that website is different.

You can keep the speeds higher by lowering the "Flow Equalization Ratio".  With it set to 100% you will get slowdowns where the lines are very wide.  In layer 9 the line width goes to 1.22mm in the narrow area between the holes and the outer wall.  That's why Cura is slowing down.  Your Flow Rate is already at 15.5mm³/sec and Flow Equalization Ratio will slow down the print speed so it doesn't go higher.  You might also want to enable Accel control at about 1000mm/sec².  Your Maximum Resolution of 0.4 is letting a lot of short line segments through on the curves and that can cause herky-jerky movement as well.

So I don't see a bug.  A large part of what is going on is you have that big nozzle in and your printing like it's a 0.4.
I'm inclined to remove the bug label as this appears to be related to your settings, but I'll leave it on for now so you can respond here.

AnthonyTTaylor, 2024-02-01:
Do you have a copy of the script you run, because if I slice on 4.6 these are not issues. 

I have models that were printed on older version of cura with same settings, and no issues.
This is only showing after i updated. 

[image]

GregValiant, 2024-02-02:
The variable line width of the 5.x versions meant new settings (I think a total of around 60) and consequently some getting used to (another learning curve).  Then there was a feature request to control the print speed according to the Flow Rate.  "Flow Equalization Ratio" is the answer they came up with.  In the early versions of 5.x it would only slow the print head down for fat lines like in that print.  In these later versions it will also speed the print head up for narrow lines.  With the print speed set at 75 I've seen speeds as high as 190mm/sec in my gcode.  There is a bug report here somewhere about that as well.

Unzip the file and put "Add Cura Settings" into your Configuration Folder and the "scripts" sub-folder.  It will be available with the other post processors under "Extensions" in Cura.
I designed the script to add a bunch of settings to the end of the gcode (alla PrusaSlicer).  The developers turned it down as being too hard to maintain because settings come and go.  The speed calculation is an option.
[AddCuraSettings.zip](https://github.com/Ultimaker/Cura/files/14132941/AddCuraSettings.zip)

AnthonyTTaylor, 2024-02-05:
I am confused by your comment about "Your Maximum Resolution of 0.4 is letting a lot of short line segments through on the curves and that can cause herky-jerky movement as well."
"So I don't see a bug. A large part of what is going on is you have that big nozzle in and your printing like it's a 0.4."

The Maximum resolution of 0.4 is regarding what? 
What settings do I have that makes it seem like I am printing with a 0.4 nozzle?

GregValiant, 2024-02-05:
The Maximum Resolution is a filter.  The lower the resolution setting the shorter the lines that are allowed into the gcode.  For a curve, that can mean a lot of short line segments to get from one end of the curve to the other.  Each of those line segments requires the printer/processor to calculate the move even though the move is .001mm.  (You have a lot of line segments that are that short in the gcode.)  A short move like that takes as much time for the printer/processor to calculate as a 200mm move.  A lot of very short moves can cause buffer under-run and the print head has to stop and wait for the next command to tell it where to go.  That causes blobs and jaggies.  When it occurs on an outside wall it's ugly.  The 0.8 nozzle has a radius of .4 so that is about the shortest length that you can notice it doing anything.  With a .4 nozzle and it's smaller radius, shorter lines (a higher resolution) are possible for it to actually notice.
The Flow Equalization Ratio is a new setting and it appears to me that the way it works has changed over the last couple of Cura releases.  The original feature request was to be able to set a "max flow rate" number and Cura would adjust the printing speed to insure that the flow didn't rise above that number.  The way it LOOKS like it works is to adjust the speed either up or down as required by the variable line width.  Your 1.2mm wide line going around the holes in those models causes a large slowdown of the nozzle because the flow rate would be well above the norm dictated by your regular line width and print speed.  A 0.8 or 1.0 nozzle doesn't react as quickly to pressure changes in the heat chamber like a smaller nozzle would.  They are harder to "control".
Set the "Maximum Resolution" to 0.6 or 0.8 and set the "Flow Equalization Ratio" or 50% or less.  You should see much smoother motion through the print.

If you don't enable "Accel Control" in Cura then the printer uses the setting in it's M201 register (Max Accel/Axis).  If that number is 2000 (it is on my E 3 Pro) and the Jerk is at 8, then the nozzle wants to blast off from one point, go screaming to the next point, and come to a near stop before continuing.  Your travel speed of 350 is OK (as long as your mechanicals don't complain) but on small prints like you have there, the nozzle might only get to that speed when it travels between the models.  Dropping the Travel Speed to 150 on your project actually results in a print time 11 seconds shorter than with the travel speed at 350.
You can think of Jerk as the "Slowest speed around a corner".  A higher Jerk value allows the print head to maintain a higher speed.  Too high and you get ringing in the print.  It's a question of balance.
So your settings which are appropriate for a small nozzle aren't so good with a large bore nozzle.  You cannot maintain high resolution with a large nozzle.
I print mostly PETG and often with a 0.6 nozzle because I print mostly functional items like you have there (as opposed to artsy objects).  It's more important to me that things bolt up as they should rather than trying to get a perfect finish on a 3D printed part.  I'm not saying that the part should be ugly, but I can't expect the same level of accuracy (resolution) with a .6 or .8 as I would get with a .2.  The setup in Cura should reflect that reality.

All that being said...I still use 4.13.1 for some models, and if I need to spiralize something I use the SmartAvionics fork of Cura which is also based on Cura 4.x.

AnthonyTTaylor, 2024-02-08:
I do not believe this is a buffer issue, I have an aftermarket SKR3 EZ board, and I've downloaded an older build of Cura 5.2.2 and no longer have the issue with the same settings. 

I've adjusted the settings you've spoke about, from 0 to 100 and all have the same outcome.
Random slowdowns in the print.

Even with default 0.4mm nozzle, standard settings - straight from Cura, fresh install, I see slowdowns in the G-Code when analyzed. 
This is a 100% a bug in the slicer. 

[image]
[image]

GregValiant, 2024-02-08:
I'm not saying there isn't a bug here.  The Cura team will look at this.

I'm still unable to reproduce it.  I tried to set the slice up the way your last image looks and I get Max Print Speed 80 and average Print Speed 80.  I searched the file for any other print speeds and there aren't any.  Every extrusion in the file is at 80mm/sec.
Post another project file of those parts but set up the way your last image shows and I'll take a look.

razethion, 2024-03-03:
I am getting the same issue, FLSUN v400 default profile on cura 5.6

[image]

Cura doesn't seem to think it's slowing down in the preview tab but playing back the gcode it's obvious that it is slowing down.

[image]

This causes major issues in skin texture on these high-speed printers.

[image]