Ultimaker/Cura #18205: Print head area in One at a Time is a parallelogram

Slashee-the-Cow, 2024-02-02:
### Cura Version

5.6

### Operating System

Windows 10 x64 22H2

### Printer

Ender-3 V3 SE, slightly customised profile

### Reproduction steps

Load a model (any model), set _Special Modes > Print Sequence_ to _One at a Time_.

### Actual results

Shaded area for print head is a parallelogram shape slightly offset from the model

### Expected results

Shaded area for print head allowance is roughly the shape of the object, with extra room around it to allow for the size of the print head. This shows up if I switch to the stock Ender-3 V3 SE profile, but that doesn't have the correct print head size (it's much smaller than the actual size, one of the reasons I created my own profile).

### Add your .zip and screenshots here ⬇️

Area shown when using my profile:
[image]
Area shown when using stock profile:
[image]
Machine settings for my printer (using my profile):
[image]
My printer definition from configuration profile, including instance, definition_changes and variants (for the nozzle):
[e3v3se slashee profile.zip](https://github.com/Ultimaker/Cura/files/14132762/e3v3se.slashee.profile.zip)

Also occurs in 5.7.0-alpha.0 nightly build.

HellAholic, 2024-02-02:
Hey @Slashee-the-Cow,

I tried your printer def and it works for me (I'm on 5.7.alpha) 
[image]

Is it behaving this way with any model? (might be a model related issue)
Or maybe you missed a file in the zip you provided?

Slashee-the-Cow, 2024-02-03:
Any model, from the basic:
[image]
to the slightly less basic (this is from a print I did, but for the screenshot I rotated one of the straight bits so you can see the shape changes with the rotation):
[image]
to the somewhat ridiculous:
[image]
I'll admit something I forgot to check earlier but I just did and it makes no difference: the same happens on my custom quality profiles as the stock profiles.

If you want me to I can upload my whole configuration folder, as long as you don't mind that there *might* be some swearing in the quality profile names. My E3V3SE profile inherits from the stock *creality_base* profile that comes with Cura.

HellAholic, 2024-02-03:
Configuration folder would be helpful :P then I can also see the plugins you have (in use and not in use). And makes it easier to reproduce the issue. Also please send in a project file with the issue in it, makes it a bit easier to get the profile changes, etc.
Random guess on my part:
- Somewhere in there, an offset is given to the nozzle which shifts the area

Slashee-the-Cow, 2024-02-04:
[slashee config 5.6.zip](https://github.com/Ultimaker/Cura/files/14156976/slashee.config.5.6.zip)
[corner screwing templates project.zip](https://github.com/Ultimaker/Cura/files/14156989/corner.screwing.templates.project.zip)
1. Please ignore all the detritus I leave in there. I highly doubt it's causing the problem, out of laziness I just tend to use the base folder for things like putting scripts into archives so I can upload them to the forums.
2. Don't use some of the scripts on your good prints. I tend to keep the "in development" ones mixed in with the rest.
3. I removed the "SettingsGuide2" plugin to get the file under the max size to upload.
4. I know there's some personal information that can be derived from some of the files, especially the log and config files. If you spread or misuse it in any way... let's just say cows are nice and peaceful, but you don't want to be near one if we're angry.
5. Helping people on the forums leaves a lot of crap like printer definitions and quality profiles I neither need nor want left over. I recently cleaned most of it out and made a copy of the config folder that I swap in if I'm loading something from the forums (don't worry, I've uploaded the "clean" version). It's possible there might be some leftovers but this has nothing to do with that because it was occurring long before I did that.
6. Moo. Are you happy now?

HellAholic, 2024-02-05:
Hey @Slashee-the-Cow,
I got the config and project, I can reproduce the issue you're having.

HellAholic, 2024-02-05:
Ok, so the Printhead Settings:
X min, X max, Y min, Y max
are not set properly somewhere along the process of creating the machine instance.
[image]
It shows as those values but they are overwritten somewhere.
Quick workaround until you find the source of the issue:
go to machine settings -> click on each of the input fields for those 4 (just to be sure also the Gantry field), then click on X(Width) input field -> after that simply close the Machine settings window and the Preferences window.
When you click on the build plate, the magic happens.

https://github.com/Ultimaker/Cura/assets/28710690/f01652bd-4213-422f-b4f3-145f3e24bd05

Alternatively, you can load this project file, should do the same thing.
[CE3V3SES_corner screwing templates.zip](https://github.com/Ultimaker/Cura/files/14163614/CE3V3SES_corner.screwing.templates.zip)

Slashee-the-Cow, 2024-02-05:
That works, I'll just have to get used to changing it after I open any project I've done before which has multiple meshes, because that reverts it... know what specific file it's being changed in so I can set it to read only?

HellAholic, 2024-02-05:
So inside your configuration folder
.../<username>/AppData/Roaming/cura/<version>/definition_changes

Line 25
What you have there:
`machine_head_with_fans_polygon = [[-40, 20], [-40, 20], [30, -60], [30, -60]]
`What should be there:
`machine_head_with_fans_polygon = [[-40,20], [30,20], [30,-60], [-40,-60]]
`

Slashee-the-Cow, 2024-02-05:
I'd say that whoever wrote Cura's JSON parser should have bad things happen to them, but I can't be 100% sure it's not you, so I'll just say thanks and that you can close the ticket if there's nothing more to do here.

closed 2024-02-05 by HellAholic