Ultimaker/Cura #18207: Error While Slicing

KrowTheBarbarian, 2024-02-02:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

AnyCubic Kobra Neo

### Reproduction steps

Error while slicing a model

### Actual results

Slicing stop

### Expected results

Should have finished
[cura.log](https://github.com/Ultimaker/Cura/files/14134693/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/14134707/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/14134719/cura.log)

### Add your .zip and screenshots here ⬇️

Nothing to add

Vandresc, 2024-02-02:
Could you try it again with this build: https://github.com/Ultimaker/Cura/discussions/18080 It doesn't contain a fix, but it has a lot more logging options, which will help us figure out what is going wrong. Thank you!

KrowTheBarbarian, 2024-02-03:
Still failed.

Log attached