Ultimaker/Cura #18211: I can't change the height of the printer in Cura, 5.6.0?

Ankalima, 2024-02-02:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

Hi, I can't change the height of the printer in Cura, 5.6.0? No matter what value I enter, nothing changes, height as it is now is too low, I have entered 215mm but it is lower.

### Actual results

Nothing.

### Expected results

The height should be changed to the specified value.

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2024-02-03:
Make sure you have the `Print Sequence` set to `All at Once`. It looks like you're configured for `One at a Time`, which limits the maximum build height based on your specified Gantry Height rather than the Z (Height).

Ankalima, 2024-02-03:
It didn't solve the problem, I removed all plugins and then it worked, then I loaded some plugins again but not all, so it seems that some plugins is the problem?

closed 2024-02-03 by Ankalima