Ultimaker/Cura #18213: cura messing up matterial used in  estimation by alot

enderus2137, 2024-02-02:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

creality ender 3 v3 se 0.2mm nozzle

### Reproduction steps

i tweaked some settings to make my model smoother, and it said it would use 4.37m of PETG fillament, even if its very small with 20% infill
[image]

### Actual results

in material usage estimation it says its gonna use 4.37m

### Expected results

way less filament usage

### Add your .zip and screenshots here ⬇️

[image]
[CE3V3SE_moup_titan4.0short_chinrtx.zip](https://github.com/Ultimaker/Cura/files/14143491/CE3V3SE_moup_titan4.0short_chinrtx.zip)

Asterchades, 2024-02-03:
This may not be as far off as you'd like to think it is. In fact, it shouldn't be off at all - Cura knows what each and every E value it commits to the slice, so any variation from that value in the actual amount of filament being used is going to come down to your printer's E setup (notably E-steps).

If you're comparing it to the resulting GCode file, you should be aware that Cura resets the E value on a sufficiently large file. The E values when this happens are between 4,157 and 4,159, though I'm not exactly sure of the specific trigger. As such if you check the very last E value you're going to see something around the 200 mark as this will follow a reset.

GregValiant, 2024-02-03:
To add to what Asterchades said,
The model is 9384 cubic millimeters in volume.  1.75 filament is 2.405 mm³/ mm.  Doing a little math that comes out to 3.901m of filament to fill the model 100%.
You have the Wall Count set to 19.   Very little of the model is Infill because you have the wall count so high.  The remainder of the material is in the brim and the supports.
[image]

This is 3 walls at 10% infill.  It requires 2.55 meters of filament which is about 60% of what your settings call for.
[image]

The 4.3 meters looks good for the way you have the slice set up.  If you want to use less material then set the wall count to 3 and the infill density to 10%.  If you want the print to be solid through-and-through you will end up using filament equal to the volume of the model + the volume of the supports and that's where you are right now.

I'll go ahead and remove the bug label and close this as the numbers are correct for your set up.  I hope you understand.

closed 2024-02-03 by GregValiant