Ultimaker/Cura #18215: Textured Mask fails slicing.

jorritmg, 2024-02-02:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Voron 2 300mm

### Describe your model

Textured mask

### Add your .zip here ⬇️

https://we.tl/t-qNTRCgXpEv

jorritmg, 2024-02-02:
Its a deadpool mask iwith the fabric texture on the outside and cura is not able to slice it. Is this a cura problem?

GregValiant, 2024-02-03:
Thanks for the report.
Something changed in Windows 11 and Cura doesn't seem to care for it.  The conflicting code hasn't been found yet so thanks for this example.

This is with my Ender 3 Pro active and with Windows 10 Pro.  You can see that it's a 58 hour print.  It took 18 minutes to slice.
[image]

jorritmg, 2024-02-03:
Yeah I managed to get it to slice only on 4.13 and 0.1 layer height. 0.15 or anything else would not do it. 

I put the mask(same as yours) on its back and that reduces the print time to 30h at 0.1mm layer height.

Hopefully they will find a fox for it.