Ultimaker/Cura #18219: Alternate Wall Line Direction

Really01101001, 2024-02-02:
### Is your feature request related to a problem?

Inconsistencies in wall thickness related to  speeds, accelerations and jerk (which can be next to impossible to get perfect and changes with filament type and brand). Stringing and oozing and possible related wall voids from repetitive tool path direction and especially seen with thin wall prints using Active Foaming and Foam materials.

### Describe the solution you'd like

There was an experimental feature with Cura 3.6 called Alternate Line Direction. This feature was absent from later versions from what I have seen also it only effected top and bottom layers but not walls from my experience.

Having a feature with the same name (Alternate Line Direction) that executed that exact thing for walls would be very good for numerous reasons. 

Such a feature could/would minimize stringing with materials prone to stringing and reduce voids in thin shell printing which can be a complication of stringing/another parallel problem with various materials particularly foaming materials like Colorfabbs Varioshore TPU, Foaming PLA and ASA. 

The effect of alternating wall line direction essentially evens out less than perfect acceleration and jerk settings, travel moves and issues where repetitive tool paths compound various "defects" in a tool path and materials attribute challenges. Another affect would be to balance layer to layer and internal material stresses caused by the extrusion vectors.

The solution would be to include a feature that would alternate layer to layer clockwise/counter clockwise line direction in general/more specifically to walls. 

For example for a single wall print on say layer 3 the line direction would move counter clockwise and layer 4 clockwise, layer 5 counter clockwise and so on. 

For multi wall prints the same would be sufficient but could also alternate wall to wall as well however I'm not included wall to wall line direction alternating in this feature request (at least not at this time) 

Please give this serious consideration I believe it would be a great feature. 

### Describe alternatives you've considered

As far as alternatives that I've considered, I'm not aware of any that would have the same effect.

### Affected users and/or printers

I think many people would benefit from such a feature. Including those using low viscosity materials including TPU and Foam/Foaming filament. Thin wall prints very much as well as end use parts that may have delicate features. Realistically everyone could benefit to some degree I believe from this and end use models that are needed to be strong, lightweight, and very smooth and dimensionally accurate.

Asterchades, 2024-02-03:
Does the existing `Alternate Wall Directions` setting, under Experimental, not already do exactly this? It's been present since 5.0.0 and enabled for all users and purposes since 5.1.0.

Really01101001, 2024-02-03:
Ok. I downloaded Cura 5 a little while ago when I got a new laptop and it's basically impossible to use as no desktop icon was present after install and there was no launcher (that I could find  present in any of the files in the package.) 

I've got v3.6 and 4.8 and primarily use 36 because I like the interface detail in the layer view render and average functionality better than other versions I've tried. 

I suppose development isn't backward compatible and 3.6 nobody would want to bother with. 

I'll have to see if I can reinstall 5+ to see if that works for me.