Ultimaker/Cura #18230: Extruder temperature is not working properly

jlknight1969, 2024-02-04:
### Cura Version

5.5.0

### Operating System

2024-01 Cumulative Update Preview for Windows 10 Version 22H2 for x64-based Systems (KB5034203)

### Printer

creality CR-10S5

### Reproduction steps

When I slice the model and then select print via usb the preset temperature setting remains is ignored and the extruder temp sits at zero. I either have to manually go to the control box and enter the temp I want or nothing happens. I cannot print any temperature towers as the code it sends to change the temp for each level does not function. This worked fine on my previous versions of cura.

### Actual results

Nothing the printer just sits there waiting because the hot end does not reach any temperature

### Expected results

The hot end should heat up.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

tigercjn, 2024-02-04:
I did a tower the other day, and it worked fine; I see you are still on 5.5.0; I used the 5.6.0 version, so maybe try updating.

GregValiant, 2024-02-04:
Thanks for the report.
"USB Printing" is a legacy feature and isn't maintained.  If it works with your printer that's fine.  If it doesn't then it doesn't.
I don't print over the USB but I always have the computer connected to my Ender so I can pass messages and see the responses.
I know that some computer/printer hardware combinations simply won't talk to each other.

Did you try to print the gcode from the SD card?

closed 2024-02-19 by github-actions[bot]