Ultimaker/Cura #18232: Temperature change and Z "hop" for bridging

cainzjuss, 2024-02-04:
### Is your feature request related to a problem?

When using materials with a steep viscosity curve, using static temperature settings results in a either (excellent bridging and undesirable stringing) or (bad bridging and no stringing). 
idea of the Z height manipulation request came due to a quirk a stumbled on when playing around with retraction settings with high speed and temperature. It made excellent extremely long bridge strings when z height was raised 0.5x current layer height (0.15mm at layer thickness 0.3mm) 

### Describe the solution you'd like

Settings in cura that:
-Change the temperature for bridges
-Distance or time to start changing the temperature before the bridge part starts (to avoid the print head staying still when changing temp)
-Z height change that takes affect at the start of the bridge (linear up) and smushes back down at the end connection of the bridge

in conjunction with the experimental settings in cura 5.5. It should be a full package (atleast for my printer, Anycubic Chiron)

### Describe alternatives you've considered

Using the post process script in cura to change temp based on layer.
This is undesirable because it changes the temp for the entire layer, producing stringing where none are wanted.
Lack of Z control at bridge start and end point.

### Affected users and/or printers

Users who use a filament with a steep viscosity-temperature curve. Meaning the filament turns from soft plastic that does not adhere to the layer below,  to absolute snot within a 1-2C temp delta range.
This would turn the difficulties of these types of filaments into a advantage, as i have achieved perfect bridge stings that are 20cm long that are thought as a guitar string. (This was by accident in an attempt to reduce stringing during travel moves)
eg. No/low odor fillaments like the Spectrum Fillaments PETG premium line

GregValiant, 2024-02-04:
I have written a post processor that (mostly) does this.  There is no "special z-hop", just the temperature change.  A couple of problems I ran into are:

- The print head really needs to move off to a park position.  That means oozing and a possible dry start to the first extrusion when it returns to the print.  An Extra Prime amount could fix that but how much extra prime would be a function of the material, the nozzle size, the time spent waiting for the temperature change.  It would really need to be tuned for every "bridge temperature change".
- The bridge outer and inner walls are usually single lines in the gcode and then the regular model starts again.  It's very difficult for the printer to react for a single line of gcode.
;TYPE:WALL-INNER
;BRIDGE
G1 F2100 E6.5
G1 F2250 X114.959 Y146.678 E0.43259  >>>> This is it - one line of extrusion.
G1 F2100 E-6.5
- If the temperature command is M109 R it might incur a 10 second dwell while the temperature stabilizes at the new value.  The time adds up and the nozzle can be oozing while it awaits the restart.  That takes us back to the "dry start" thing.  Using the M104 S command is an option, but unless the bridged area is large the hot end just doesn't react fast enough.

Here are what I found to be two problem areas with the inner and outer bridge walls.  There are two more on the other side.
[image]

I made it an option to ignore the bridge walls, but since they are integral to the whole "Bridge" thing I wasn't happy with that.  I've never used the script as it remains clumsy.  Adding the "1/2" layer z-hop" wouldn't be difficult.
[image]

As an alternative I wrote a different script "Support-Interface Material Change" that pauses and allows the user to change material for the Support Interfaces and then change back to the model material when the interface ends.  That's the one I use and it works really well on largish flat areas.  For things like horizontal holes - you can introduce so many pauses/filament changes that it gets pretty annoying pretty fast.  Using disparate materials (PETG for the interface on a PLA print) does allow the supports to come right off and the finish is pretty darn good with no scarring.  I like that a lot.  Another plus is that the airgap over the interface can be zero.  That means there is no dimensional problem incurred as the first-layer-over-interface can't sag.

cainzjuss, 2024-02-05:
Interesting

One way to combat the ooze might be to slow down the print before the bridge in order to have plenty of time for the temperature to change, without moveing the head to the side. And make other build up changes required for different materials (judging by your screenshot, your material behaves vastly different to mine). Maybe even, if there is a long linear move before the bridge (making it impossible to inject commands), chop that up into multiple commands in order to be able to slide in commands at the right'ish point in time. 
Theoretical:
   Print temperature: 210C
   Bridge temperature: 250C
   Print speed: 40mm/s
   Bridge speed: 100mm/s
   Print head temp change (up): 5C/s
   Print head temp change (down): 10C/s
This would mean in order for the temp to be up in time for the bridge, it would need to put the temp command 32cm before the bridge starts. However, if the speed were to be dropped down to 5mm/s, it would only require 4cm of distance to reach the required temp. If down to 3mm/s it would take 2,4cm. There might be a slightly different appearance due to changing temperature and nozzle backpressure (if such a thing exists) but i think its worth it for the 1-2 layers near the bridge. Maybe can be combated with a slight retraction?

(My Gcode is rusty so here is my pseudo code for the theoretical)
...
G1 XYE normal 
... 
;(4cm before bridge)
Temp change command
Drop speed command
G1 XYE slow
;Bridge starts
Bridge speed command
G1 Z command (if any)
G1 XYE (Z?) command
G1 Z command (if any)
if bridge continues or next part is close, continue with bridge settings, if not ... slow down and back to normal settings
 

I found these settings in cura.  Where can your script be found so i can try and smash the 2 together?
[image]