Ultimaker/Cura #18248: Simple part slicing failed.

KartrH, 2024-02-06:
### Cura Version

5.6.0

### Operating System

Windows 11 (up to date including AMD drivers)

### Printer

Creality Ender 3

### Name abnormal settings

.75 retraction (MicroSwill NG extruder)

### Describe model location

Middle of the plate.  When I changed support from "Touching Bed" to "Everywhere" it sliced

### Describe your model

Simple model, have sliced before, but with different extruder. (Several Cura Versions ago)

### Add your .zip here ⬇️

[Snap On Soldering Clamp (Helping Hands) - 6036321.zip](https://github.com/Ultimaker/Cura/files/14173440/Snap.On.Soldering.Clamp.Helping.Hands.-.6036321.zip)

HellAholic, 2024-02-09:
Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.
If you could also share the project file instead of the STL/Step files, that would help narrow things down.

KartrH, 2024-02-09:
Tonight I will try.  Honestly, I will have to look up my original email to
see which part it was.  I'm fairly sure I uploaded the part through the
submission.

(I know how frustrating it is trying to troubleshoot without details).

KartrH, 2024-02-09:
The zip file is linked in the report on GitHub.  Also, I tried a different
copy of the file and tried unzipping using 7zip and the default
decompression utility in Windows 11.