Ultimaker/Cura #18250: Bug slicing a plate negative

preminedor, 2024-02-06:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

ender 5

### Describe model location

the concave part is facing up and the plane one is touching the buildplate

### Describe your model

the negative of a plate (a few concentric circles)

### Add your .zip here ⬇️

[Plato (Molde B1).zip](https://github.com/Ultimaker/Cura/files/14178209/Plato.Molde.B1.zip)

HellAholic, 2024-02-09:
Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc