Ultimaker/Cura #18254: Slice failure

wprjrdev, 2024-02-06:
### Cura Version

5.6.0

### Operating System

MacOS Sonoma 14.2.1

### Printer

Ender 3 V2

### Name abnormal settings

No. Yes.

### Describe model location

Yes

### Describe your model

No. Yes. Yes.

### Add your .zip here ⬇️

I have been having issues with multiple files since yesterday. Prior to that everything was working fine. Attempted to fix by uninstalling/reinstalling 5.6.
[Remix Nishihara Infini Cutting Mat Holder.3mf.zip](https://github.com/Ultimaker/Cura/files/14183007/Remix.Nishihara.Infini.Cutting.Mat.Holder.3mf.zip)

wprjrdev, 2024-02-06:
I just went into the **Build Plate Adhesion** section, selected **None**, and attempted to slice again. The slice was successful this time in the position (upright) it was in when I got the slice failure notice.

HellAholic, 2024-02-09:
Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc