Ultimaker/Cura #18255: Slicing of lattice plates using the Fine resolution failed

dianaragut, 2024-02-06:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ultimaker 3

### Name abnormal settings

Just took off adhesion

### Describe model location

No 

### Describe your model

Two lattice plates, I could slice the model using the Normal resolution instead of fine.

### Add your .zip here ⬇️

[UM3_L5T4A15.zip](https://github.com/Ultimaker/Cura/files/14184608/UM3_L5T4A15.zip)

HellAholic, 2024-02-09:
Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc