Ultimaker/Cura #18256: Slicing Error

C1PH3R-3RR0R, 2024-02-06:
### Cura Version

5.5

### Operating System

Windows 10

### Printer

Creality Ender3 V2

### Name abnormal settings

It crash with generated support

### Describe model location

move it away from the center of the buildplate it slice

### Describe your model

if you get 90% it slice also

### Add your .zip here ⬇️

[CE3E3V2_body.zip](https://github.com/Ultimaker/Cura/files/14184629/CE3E3V2_body.zip)

HellAholic, 2024-02-09:
We have a ticket with the aim to resolve/improve the behavior. It will be included in the next cura release.
For internal references CURA-11603

Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc