Ultimaker/Cura #18260: Unable to slice with tree as support type

TheBarber82, 2024-02-07:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Sidewinder x1

### Name abnormal settings

support type as tree

### Describe your model

Yes without normal support

### Add your .zip here ⬇️

[ASW_X2_RIGHT_Z_SUPPORT.zip](https://github.com/Ultimaker/Cura/files/14187819/ASW_X2_RIGHT_Z_SUPPORT.zip)

HellAholic, 2024-02-09:
We have a ticket with the aim to resolve/improve the behavior. It will be included in the next cura release.
For internal references CURA-11603

Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.

TheBarber82, 2024-02-09:
Hi, the build 18080 seems to works well GG.

closed 2024-02-09 by TheBarber82