Ultimaker/Cura #18263: Cura 5.5.0 - Z slic error final print

atamico, 2024-02-07:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

ANET A8 - Marlin firmware 2.1.2.1

### Reproduction steps

I drew a hollow cube with the measurements:
20mm - length
20mm - width
10mm - height

### Actual results

The post-print result was:
20mm - length: OK
20mm - Width: OK
8.5 - Height - THE PROBLEM IDENTIFIED

### Expected results

The Z axis step check has already been checked with a caliper and no problem was identified with manual movement of the axis. I believe that the layer/line calculation is being calculated incorrectly by the slicer.

### Add your .zip and screenshots here ⬇️

[Cura 5.5.0 sli z error.zip](https://github.com/Ultimaker/Cura/files/14200064/Cura.5.5.0.sli.z.error.zip)

[cura.log](https://github.com/Ultimaker/Cura/files/14200081/cura.log)

HellAholic, 2024-02-09:
You probably need to check the shrinkage for the material you're using. I suspect you are using ABS, but check the specifications of the material for shrinkage factor.
If you also share a project file instead of Gcode/STL we can help you better.

atamico, 2024-02-09:
> You probably need to check the shrinkage for the material you're using. I suspect you are using ABS, but check the specifications of the material for shrinkage factor. If you also share a project file instead of Gcode/STL we can help you better.

Other tests were carried out, using other slicers and also other larger models. A difference of 1.5 mm in height was present in all cases (regardless of the size of the piece). So what really happens is a question of the material being used. and yes it is/was ABS.

Thanks for the tip and suggestion

closed 2024-02-09 by atamico