Ultimaker/Cura #18264: duplicated stl objects slicing failed

hmt0934, 2024-02-07:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

ArtillerySidewinderX2

### Name abnormal settings

tree support, 60% infill

### Describe model location

centered, duplicated, rotated 90 degree on X

### Describe your model

yes, sliced; not watertight; no

### Add your .zip here ⬇️

[ASX1_FailedSliceSTL.zip](https://github.com/Ultimaker/Cura/files/14202383/ASX1_FailedSliceSTL.zip)

HellAholic, 2024-02-09:
Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.

hmt0934, 2024-02-12:
I don't want to install private builds, I'd prefer waiting for a stable Cura version.
The 5.6 release is using too much system resources, CPU fan starts directly when Cura 5.6 initializes, like a Steam FPS game.
There is also a program exit-cache cleaning issue, it has to be investigated. I had to clean the Cura cache directory two times in 3 weeks.

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant