Ultimaker/Cura #18265: Slicing Failed

reboot689, 2024-02-08:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3 V3 SE

### Name abnormal settings

not sure? it prints anything else just fine. You don't yet have "Default" profiles for the E3V3SE yet

### Describe model location

Centered, rotated 45 degrees to fit

### Describe your model

I have not. it IS but it's a skeleton. not exactly hole free? Mesh tools crashed the Program

### Add your .zip here ⬇️

https://we.tl/t-YAIHuk1Mg1

HellAholic, 2024-02-09:
Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc