Ultimaker/Cura #18267: Нарезка на слои не выполнена из-за непредвиденной ошибки.

7ARTA7, 2024-02-08:
### Cura Version

5.5.0

### Operating System

Windows 10 PRO

### Printer

Anycubic Vyper

### Describe your model

Moon

### Add your .zip here ⬇️

[Uploading AV_moon.zip…]()
Moon

HellAholic, 2024-02-09:
Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc