Ultimaker/Cura #18271: Tootless model crash

mewbusi, 2024-02-08:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3

### Name abnormal settings

Switching to dynamic quality default still fails.

### Describe model location

Turning it 90 degs verticial does seem to stop the crash

### Describe your model

A small toothless figurine. It has sliced before

### Add your .zip here ⬇️

[CE3E3V2_toothless.zip](https://github.com/Ultimaker/Cura/files/14214956/CE3E3V2_toothless.zip)

HellAholic, 2024-02-09:
Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.

mewbusi, 2024-02-09:
> Sure, I can take a look sometime today.

I have gone ahead and installed the alpha. Slicing still fails as expected,
hopefully you should be able to get the data from its logging.

HellAholic, 2024-02-16:
Sorry it took me a while to get back to you.
Small update, we're tracking the issue.
For internal references
CURA-11569