Ultimaker/Cura #18276: More options for controling smoothing spiralized contours at beginnign and at the end of spiralized countur.

bartwaw, 2024-02-09:
### Is your feature request related to a problem?

printing in vase mode with bigger nozzle (in my case 1mm )  introduce  still visible squishing of first layer in vase mode and specially at the end of spiralized countour.
Using option "smmothing" slighlty solves problem on bottom but top part we stil have visible squshing.

### Describe the solution you'd like

I think that we should get more variables (options) to control smoothing option for bottom and top part of spiralized countour.

### Describe alternatives you've considered

I think that we should have control (of smmothing part):
length
flow
line width

### Affected users and/or printers

It will improve vase mode and specially for bigger diameter nozzles. I have on the way 1.5mm nozzle and thint that squshing will be even more visible.

smartavionics, 2024-02-09:
My Cura variant tapers the flow at the start and end of the spiral. See https://github.com/smartavionics/Cura/releases.

bartwaw, 2024-02-09:
yup, tested. It behaves on top and bottom ?

GregValiant, 2026-10-01:
UM Cura was adjusted to follow the SmartAvionics lead and now tapers the first spiral layer and the top spiral layer.
I'll close this as stale.

closed 2026-10-01 by GregValiant