Ultimaker/Cura #18277: Possible bug in vase mode - strange even layers

bartwaw, 2024-02-09:
### Cura Version

every

### Operating System

Windows

### Printer

every

### Reproduction steps

Vase mode generates additional lines in bottom layers - at every second layer.

If we setup vase mode to more bottom layers than one we get every odd layer with uncontrolable look behavior

### Actual results

So first layer is like we set in slicer:
[image]

second one looks like this:
[image]

third again normal:
[image]

4th same as 3th:
[image]

### Expected results

even layers should be same as odd layers

### Add your .zip and screenshots here ⬇️

[CFFFP_Part5.zip](https://github.com/Ultimaker/Cura/files/14218565/CFFFP_Part5.zip)

smartavionics, 2024-02-09:
That's how it has always behaved when spiralizing. It's a feature, not a bug. I forget the reasoning behind the extra walls.

bartwaw, 2024-02-09:
Hi, i know but that,  tested your fork also. 
but personally think that this feature should be turned off. It's does nothing to improve anything. If you want to print with 4 bottom layers and specially with bigger nozzle, then bottom layer looks reallt strange. There should be option to use standard infills.

smartavionics, 2024-02-09:
I'll add an option to my Cura to disable the extra walls. It will be available in the next release.

bartwaw, 2024-02-09:
Great, thanks.

GregValiant, 2024-02-09:
This is expected behavior so I'll remove the bug label.
@bartwaw  You could put in a feature request here for UM Cura to incorporate the same sort of "switch" that SmartAvionics is going to put into his fork.
I'm going to close this report.  Let me know if you feel that is premature.

closed 2024-02-09 by GregValiant