Ultimaker/Cura #18279: Unable to Open File, The file could be corrupt, inaccessible or it did not contain any models.

he60, 2024-02-09:
### Cura Version

5.6

### Operating System

WIndows 10

### Printer

Ender 3 pro

### Reproduction steps

Load the 3d file

### Actual results

Unable to Open File, The file could be corrupt, inaccessible or it did not contain any models.

### Expected results

Load the 3d file

### Add your .zip and screenshots here ⬇️

Cura 5.6
[image]

Old Cura 4.13
[image]

Logs Cura 5.6
[cura 5.6.log](https://github.com/Ultimaker/Cura/files/14223018/cura.5.6.log)

Logs Cura 4.13
[cura 4.13.log](https://github.com/Ultimaker/Cura/files/14223019/cura.4.13.log)

The 3D file in question & project file
https://drive.google.com/file/d/1cVF83hOqemAveRMz6Zg_2955_VjEEiBo/view?usp=drive_link

he60, 2024-02-09:
Edit, It works when opening the 3d project, so it can be opened if i load it in Cura 4, then saving the project, then open it on cura 5

nallath, 2024-02-12:
The STL is invalid. It seems that the old implementation incorrectly allowed you to load it. One of the values in the STL is "-1.#ind" which isn't a valid number.

closed 2024-02-12 by nallath