Ultimaker/Cura #18280: UltiMaker Cura is not a proper AppImage

HarlemSquirrel, 2024-02-09:
### Cura Version

5.6.0

### Operating System

Pop!_OS 22.04

### Printer

Lulzbot TAZ 6

### Reproduction steps

1. Use [appimaged](https://github.com/probonopd/go-appimage) to manage AppImages
2. Place Cura AppImage in `~/Applications/`
3. Observe appimaged set the system shortcut for Cura
4. Launch Cura from system applications

### Actual results

See system notice
> UltiMaker Cura is not a proper AppImage
> Too short Please ask author to fix it

### Expected results

No system notice

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/14223154/cura.log)

Steve-Tech, 2024-02-12:
I get the same message also, on Debian Testing with appimaged 817. Cura does run fine though.

nallath, 2024-02-12:
Uh. That's a very... helpfull message from the OS. Is there any way that it can tell us _why_ it believes it to be an invalid appimage?

HarlemSquirrel, 2024-02-12:
Looks like the message is coming from here:

https://github.com/probonopd/go-appimage/blob/09fd0186774aefa2351c42b4bb22f92ce0c4f235/src/appimaged/appwrapper.go#L35-L49

Steve-Tech, 2024-02-13:
This file seems to throw the error: https://github.com/probonopd/go-appimage/blob/09fd0186774aefa2351c42b4bb22f92ce0c4f235/internal/helpers/updateinformation.go

It seems to be related to update information, so yeah it has no impact on usability, but also seems like a simple fix (but I have no experience with building AppImages).

Steve-Tech, 2024-02-13:
So the `.upd_info` section seems to contain `guess` while other appimages leave it blank. According to the [AppImage spec](https://github.com/AppImage/AppImageSpec/blob/master/draft.md#update-information) the correct options are `zsync`, `gh-releases-zsync`, `pling-v1-zsync`, or `bintray-zsync`.
```sh
stephen@Stephens-PC:~/Applications$ readelf -p .upd_info UltiMaker-Cura-5.6.0-linux-X64.AppImage

String dump of section '.upd_info':
[     0]  guess

stephen@Stephens-PC:~/Applications$ readelf -p .upd_info balenaEtcher-1.7.9-x64.AppImage

String dump of section '.upd_info':
No strings found in this section.
```

Although I feel like appimaged should skip validation when the method isn't supported, unless `guess` is documented somewhere, I think the bug is in Cura.

Steve-Tech, 2024-02-13:
GitHub's search reveals this line to be the likely problem: https://github.com/Ultimaker/Cura/blob/234a8115b9a710943045de3fd31a1a13d9d95224/packaging/AppImage-builder/AppImageBuilder.yml.jinja#L78

I'd recommend removing it, or setting it to `""`, depending on if it's actually required or not. Or alternatively implementing the creation of a .zsync file and using something like `gh-releases-zsync|Ultimaker|Cura|latest|UltiMaker-Cura-*-linux-X64.AppImage.zsync`

GregValiant, 2026-10-01:
I'll close this as stale.

Steve-Tech, 2026-10-01:
The issue is kinda still there, and something like this should fix it:

```diff
--- a/packaging/AppImage-builder/AppImageBuilder.yml.jinja
+++ b/packaging/AppImage-builder/AppImageBuilder.yml.jinja
@@ -75,4 +75,3 @@
 AppImage:
   arch: {{ arch }}
   file_name: {{ file_name }}
-  update-information: guess
```

I was just hoping someone more familiar with the codebase could figure out the zsync stuff instead.

This also seems like a relevant issue: https://github.com/probonopd/go-appimage/issues/276

GregValiant, 2026-10-01:
@HellAholic ?

HellAholic, 2026-10-01:
We're aware of a few issues affecting the AppImage, but unfortunately, we haven't had the time to investigate them and make validated fixes (with testing and all).
There is this PR https://github.com/Ultimaker/Cura/pull/21823 (draft - also unclear if it gets to 5.14). So should be reference for the issues at least.

CURA-13360

rephlex, 2026-10-25:
Pop!_OS 22.04 is jammy-based, and for that base there is a proper .deb now: my company builds Cura (5.9.1 for 22.04, 5.13.0 for newer bases) in a signed community APT repo. apt install, menu entry, file associations, updates included, no appimaged needed:

```
curl -fsSL https://mirror.plustech.de/plustech.gpg | gpg --dearmor -o /usr/share/keyrings/plustech.gpg
echo "deb [signed-by=/usr/share/keyrings/plustech.gpg] https://mirror.plustech.de/ubuntu/ plustech-jammy main" > /etc/apt/sources.list.d/plustech.list
apt-get update && apt-get install ultimaker-cura
```

https://mirror.plustech.de/ , sources published, community project.