Ultimaker/Cura #18282: Tree supports not working

Doinikdrnz, 2024-02-09:
### Cura Version

elegoo cura 5.6.0

### Operating System

Windows 11

### Printer

elegoo neptune 4 pro

### Name abnormal settings

chanched from normal supports to tree supports

### Describe model location

center

### Describe your model

yes i have it stopped working as i turned on tree supports

### Add your .zip here ⬇️

[EN4PRO_safe_handle.zip](https://github.com/Ultimaker/Cura/files/14225656/EN4PRO_safe_handle.zip)

WhatNot911, 2024-02-10:
Same issue as mine
[#17916](https://github.com/Ultimaker/Cura/issues/17916)

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc