Ultimaker/Cura #18285: Feature request: More hight when printing objects in sequence.

HMarxen, 2024-02-10:
### Is your feature request related to a problem?

When printing objects in sequence, the height is limited to your printer's distance between the nozzle and gantry. 

### Describe the solution you'd like

For bed slingers and core xy printers it would definitely be possible to print higher objects in sequence, they just have to be oriented behind each other on the print bed.
I would like to have a sequence print mode where I can print higher objects as long as I position them along the y axis. Cura could block all the area to the left and right of my first placed object and only allow me to place the second object in front or behind the first.

### Describe alternatives you've considered

None

### Affected users and/or printers

All bed slingers and core xy

GregValiant, 2024-02-13:
The gantry height can be changed.  It's in the Machine Settings.  You can make it whatever you want.
[image]

The setting is there as a safety.  It is certainly possible to print taller models as long as the user pays attention and checks all the movements and clearances two or three times to make sure.  Not many people do that.
I really don't see the Cura team altering that situation since the gantry height is specified by each printer definition author for each particular printer.

GregValiant, 2026-10-01:
I'll close this as "won't do".

closed 2026-10-01 by GregValiant