Ultimaker/Cura #18288: printspeed of a windshield

Jogy-s, 2024-02-10:
### Is your feature request related to a problem?

Description: If you print a windshield, it is always printed with speed of the brim (15mm/s in my case).

### Describe the solution you'd like

It would be fine if I can set the speed of printing the windshield.
Either I can't find it, or you can't set it.

### Describe alternatives you've considered

I don't know. How can I set the printspeed of windshield.

### Affected users and/or printers

everyone

GregValiant, 2024-02-12:
@Jogy-s are you talking about a draft shield?  The "Skirt/Brim" speed is the speed for both a draft shield and for an ooze shield as well as for any Skirt or Brim.

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant