Ultimaker/Cura #18289: When scaling up, errors out

matthewjt, 2024-02-10:
### Cura Version

5.7.0-alpha.0

### Operating System

Windows 10

### Printer

AnyCubic Kobra Neo

### Describe model location

It slices if I have it less than 115% scale

### Describe your model

Export from WoW Model Maker

### Add your .zip here ⬇️

[CFFFP_Mattiam-Ready v2_115 percent.zip](https://github.com/Ultimaker/Cura/files/14229610/CFFFP_Mattiam-Ready.v2_115.percent.zip)
It's very odd because it'll slice if I'm at 110% scale, but at 115% it errors out.

matthewjt, 2024-02-10:
Another oddity, if I import into Tinkercad then scale up (I did 125%) then slice, it works fine
[Mattiam-Ready.zip](https://github.com/Ultimaker/Cura/files/14229617/Mattiam-Ready.zip)

matthewjt, 2024-02-10:
[Mattiam-Ready v2.zip](https://github.com/Ultimaker/Cura/files/14229621/Mattiam-Ready.v2.zip)
Oops, uploaded the wrong STL.  This ZIP has the original then the one scaled up in Tinkercad

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc