Ultimaker/Cura #18293: Concentric ironing pattern leaves areas unironed

Gerhard-59, 2024-02-11:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Anycubic Kobra

### Reproduction steps

Sliced a coin shaped thing with "Ironing pattern - Concentric" enabled

### Actual results

4 areas of the surface haven't been ironed. See enclosed screenshot - marked with blue circles

### Expected results

The complete surface sould be ironed with concentric lines.

### Add your .zip and screenshots here ⬇️

[PET_Coin-for-4mm-PTFE_circular.zip](https://github.com/Ultimaker/Cura/files/14231411/PET_Coin-for-4mm-PTFE_circular.zip)
[image]

GregValiant, 2024-02-12:
Thanks for the report.
This is a zoom view of your screenshot.  The ironing settings are "0.1 Line Width" and "10% Flow".  The vertical yellow lines (I added the black lines to illustrate the area I'm referring to.) look like combing moves to me and the preview shows them as extrusions.  They are very narrow because of the settings.
Additionally, I read the gcode into AutoCad and the lines are in there as extrusions.

[image]

So I can't duplicate a problem with your project file.  If you have a gcode that you believe is missing the moves from that area please post it.  Right now I don't see a bug here.

Gerhard-59, 2024-02-12:
Hello Greg,

Thank you for your fast answer. I replayed the extrusion in Cura an noticed that the lines were extruded. It seems that I have been a victim of an optical deception. I'm sorry for the confusion.
Regards Gerhard

GregValiant, 2024-02-13:
Not a problem.
I don't use ironing much.  I did notice that with the flow at 10% my nozzle starts to push a ridge of plastic around.  At 5% flow it's a lot better.  So depending on the printer, it can benefit from some tuning.

closed 2024-02-13 by GregValiant