Ultimaker/Cura #18296: Cant read 3mf files

ljohnsonnsb, 2024-02-11:
### Cura Version

5.6

### Operating System

Windows 11

### Printer

Custom printer or any printer

### Reproduction steps

I save a project, it is saved as a 3mf file. 
When I try to load the file I get an error saying it is an 'Invalid File' . If I look at all the file types in the 'open' window 3mf is not listed so I have to choose 'all files'. 

### Actual results

I get an error saying 'Invalid File"

### Expected results

Load the file.

### Add your .zip and screenshots here ⬇️

Log and 3mf files:
[Cura_fail.zip](https://github.com/Ultimaker/Cura/files/14232816/Cura_fail.zip)

nallath, 2024-02-12:
You have disabled the 3MF reader. You need to enable it again in the marketplace.

closed 2024-02-12 by nallath

ljohnsonnsb, 2024-02-12:
@nallath I have NOT disabled 3MF reader. I went through the all of the marketplace apps and '3MF Reader' is not there. I'm curious why you closed this when it hasn't been resolved?