Ultimaker/Cura #18299: Slicing failed trying to slice half of model Sharlin_Warcruiser_1.stl

collettm, 2024-02-12:
### Cura Version

5.5.0

### Operating System

Mac OS Monterey

### Printer

Creality Ender 3 S1 Plus

### Describe model location

Model is scaled and move d into print area, rotated 90 degrees, and sunk by half the z value to print a half of the model.

### Describe your model

Model has been sliced whole when rotated to point upwards

### Add your .zip here ⬇️

Location of file here:
https://www.thingiverse.com/thing:4284725
Both versions fail with the same slicing error. An additional model like it at 
https://www.thingiverse.com/thing:3784126
fails as well. Reason for attempt to slice in this manner was to print two halves of model to then glue together.
[CE3S1P_Sharlin_Warcruiser_1_Slicing Fail.3mf.zip](https://github.com/Ultimaker/Cura/files/14234972/CE3S1P_Sharlin_Warcruiser_1_Slicing.Fail.3mf.zip)
I have printed a scaled and rotated version where the ship points upward, though the print result was poor and required a lot of support.

GregValiant, 2024-02-12:
Thanks for the report.
That model has a LOT of errors.
[image]

When a model has errors then whether or not it will successfully slice can depend on the orientation.  When you drop half the model below the build plate a lot of those errors will leave inverted surfaces and open seams.
After repairing the model (MS 3D Builder) it sliced without issue.
[image]

Let us know if a repaired model will slice for you.  I'll remove the bug label in the meantime.

Here is a report on the errors as found by https://formware.co/OnlineStlRepair
--> 3405 Naked edges (?)
--> 71 Planar holes (?)
--> 103 Non-planar holes (?)
--> 266 Non-manifold edges (?)
--> 77 Inverted faces (?)
--> 62 Degenerate faces (?)
--> 4 Duplicate faces (?)
--> 0 Disjoint shells (?)

GregValiant, 2024-02-13:
I only know Windows.  Here is the model repaired in Builder.
[MS3D_Sharlin_Warcruiser_1_Slicing Fail.zip](https://github.com/Ultimaker/Cura/files/14270384/MS3D_Sharlin_Warcruiser_1_Slicing.Fail.zip)

Don't tell anyone where you got it.  I don't do peoples homework for them but you made an effort and I'll make an exception.
When I split a print I make 2.1 diameter pin holes in the halves.  Drop two pieces of filament into one half and you have locating pins for when you bond the halves together.  The halves will line pretty near perfectly.
BTW, I also sliced it with a 0.3 nozzle.  That model is fine enough to benefit from the narrower line width and a 0.1 layer height.  (I didn't drop to the 0.2 because they don't work with my Ender.)
It looked a lot nicer with the smaller nozzle.  Some of the antenna/probe things are still partials and won't make it.  It's a tough print.

closed 2024-02-13 by GregValiant

collettm, 2024-02-14:
One final note of thanks; as a noob when it comes to Cura and fixing STLs - as well as a first time printer owner - this has been great to learn from. Thanks for the added comments / tips on split printing too. I may try printing on .12 resolution on my Ender 3 S1 Plus, given the model resolution. Take Care, m.