Ultimaker/Cura #18305: Merging modles move them on the build plate

CoreyBurger, 2024-02-12:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 5 with Microswiss Direct Drive

### Reproduction steps

Select model 1 and model 2
Right click
Choose Merge Models

### Actual results

Both models jump to a new location

### Expected results

The models merge in place without jumping location

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

fieldOfView, 2024-02-12:
> Expected results
> The models merge in place without jumping location

Then group them instead.

Merging the models explicitly moves each model so that their original origins coincide (ie the origin defined in the file, not shown to the user in Cura). The intended usecase is that you design a model in two parts, export them as separate STL files, and then need to get them to be in the exact same relative position as they were in the CAD package you used to design the model with.

In cura "merge" == "moved together and group". So if you don't want the models to move, then just "group" instead.

I am closing this issue, since the behavior is by design.

closed 2024-02-12 by fieldOfView

CoreyBurger, 2024-02-12:
Ahh, ok. And merge isn't doing what I except it to either - which is treat as a "single object and slice accordingly". I would suggest that "merge" is the wrong choice of word for what is actually "centre and group"